cornell-brg/pymtl #118: Connect error message does not report which line in model caused error

cbatten, 2015-02-05:
PyMTL throws a ConnectError when there is a bitwidth mismatch between two ports being connected together, but it does not say what line in the model caused the error. This seems like a pretty big issue?

dmlockhart, 2015-02-05:
Do you have a suggestion on how to fix this?

I don't think there's any way for the ConnectError exception to know what line the error occurred on, this information is only known further up the stack trace.  The stack trace does indeed report this information, the real complaint is that the stack trace is too long and the relevant information requires scrolling up.

The only way I can think to reduce the stack trace is to catch the ConnectError in a try/except block and keep reraising it until it reaches  s.connect or s.connect_dict... this seems really unclean.

dmlockhart, 2015-02-05:
Actually, I think I might have a good way to do this... I didn't realize I could just catch the Exception using try/except within s.connect

commit b441ffa8 (2015-02-05): [pymtl] better error msgs for connect (fixes #118)
closed 2015-02-05 by dmlockhart