cornell-brg/pymtl #144: connecting wire slice to wire slice seems not to work

cbatten, 2015-10-30:
See:

  * https://github.com/cbatten/pymtl-wire-to-wire

PyMTL doesn't complain with connecting bitslices of wires, but even so there seems to be a deeper issue. If you take this line:

```
  s.connect( s.wire2[0], s.wire1[0] )
```

and change it to this:

```
  s.connect( s.wire2[0], s.wire1[0] )
```

It works. PyMTL creates implicit combinational blocks to handle slicing -- it needs to know the src/dest in order to do that, and I wonder with wires if it assumes the first argument is the input and the second one is the destination?

dmlockhart, 2015-10-30:
Is this related to connect_wire()?

That was added because connecting two wires did not provide enough directionality information for translation to work properly.

Is simulation broken in this case?

cbatten, 2015-10-30:
It is kind of related to connect_wire -- I thought connect_wire was just for translation -- we needed to know the direction to correctly to Verilog translation, but I did not think connecting wire to another wire was a simulation issue -- and it isn't -- the real issue is connecting a wire bit slice to a wire bit slice. If you don't put those in the right order (either src then dest with connect or dest then src with connect_wire) you get incorrect simulation behavior. So I am not quite sure what the right approach is -- maybe to detect the case where we try to connect two wire bit slices and force the user to use connect_wire -- and hope the user reads the error message to know that connect_wire only works if you do dest then src.