cornell-brg/pymtl #150: Cannot wrap a VerilogModel into a pymtl model for simulation

jsn1993, 2016-06-01:
For example, I want to import module VvaddXcel from VvaddXcel.v, and then use a PyMTL model VvaddXcelHLS that defines s.vvadd = VvaddXcel(), for simulation.

It turns out that I cannot define a combinational block that communicates with s.vvadd. These are the things I put in the combinational block

@s.combinational
def comb_logic( s )
      s.vvadd.xcelreq.msg.value = ~s.xcelreq.msg
      s.vvadd.xcelreq.val.value = not s.xcelreq.val
      s.vvadd.xcelreq.rdy.value = s.xcelreq.rdy

The error:
/home/sj634/Installations/pymtl/pymtl/tools/ast_helpers.py:115: in get_closure_dict
    closure_objects = [c.cell_contents for c in fn.func_closure]
E   TypeError: 'NoneType' object is not iterable

I debugged for some time, and found that the closure dict somehow stores the modules/ports that one decorated combinational block/posedge_clk block communicates with. (Correct me if I'm wrong here).

When I remove the combinational block in VvaddXcelHLS, pymtl doesn't throw error message. But obviously I need to do something in the combinational block for correctness.

hawajkm, 2016-06-01:
Do you have an example that I can use to reproduce the error?

jsn1993, 2016-06-01:
Sadly it's not easy to get a simple example working in a minute.

closed 2016-06-01 by jsn1993

hawajkm, 2016-06-01:
This what I can give, then:

Try connecting some stuff directly:
`s.connect( s.vvadd.xcelreq.msg, s.xcelreq.msg)
s.connect( s.vvadd.xcelreq.rdy, s.xcelreq.rdy)`

For the rest, where you need some logic, try creating a wire:
`s.xcelreq_notval = Wire(1)
s.connect( s.xcelreq_notval, s.vvadd.xcelreq.val )

def comb_logic( s )
  s.xcelreq_notval = ~s.xcelreq.val`

P.S.
I am not sure whether you actually wanted to invert the complete message or that was a typo.

jsn1993, 2016-06-01:
My bad,   def comb_logic( s ) shouldn't contain the parameter s!