cornell-brg/pymtl #168: Use apt-get install flex-old instead of flex

wizard97, 2018-10-19:
Verilator fails to compile with only the "flex" package due to a missing `FileLexer.h`. You can see this header is no longer part of the flex package (https://packages.ubuntu.com/bionic/amd64/flex/filelist).

Instead the README should be updated to use flex-old, which has this header (https://packages.ubuntu.com/bionic/amd64/flex-old/filelist).

cbatten, 2018-10-19:
great! thanks for the feedback. We are working on a new version of PyMTL which will use the latest version of Verilator too ...

cbatten, 2018-10-19:
README updated!

closed 2018-10-19 by cbatten

cbatten, 2019-01-22:
Thanks! So do we need to replace this:

```
 % sudo apt-get install git make autoconf g++ flex bison
```

with this:

```
% sudo apt-get install git make autoconf g++ libfl-dev bison
```

wizard97, 2019-01-22:
Yep, that worked for me with the latest verilator!