cornell-brg/pymtl #182: pymtl doesn't correctly register combinational blocks that write to slices of an OutPort

Xaec6, 2019-09-23:
For example, the following code does not guarantee that models wired to the outport will execute after the combinational block is finished, but adding `s.out.value = 0` at the beginning of the block fixes the issue.

    class BuggyClass( Model ):    
        def __init__( s):
            s.in_ = InPort ( 4 )
            s.out = OutPort( 2 )
        
            @s.combinational
            def logic():
                s.out[1] = ~s.in_[1]
                s.out[0] = s.in_[3]

yo96, 2019-09-23:
You should use `.value` Inside your combinational block, i.e.:

```
        @s.combinational
        def logic():
            s.out[1].value = ~s.in_[1]
            s.out[0].value = s.in_[3]
```

Hope this solves your problem!

Xaec6, 2019-09-23:
oh i didn't realize value could be used on slices, thanks!

closed 2019-09-23 by Xaec6