cornell-brg/pymtl #183: possible bug in pclib/ifcs/ValRdyBundle.py

egaburov, 2019-10-25:
Hi,

Was going through PymTL Tutorial3, and noticed something funny  on Fig.41 and it is the same i the code: https://github.com/cornell-brg/pymtl/blob/master/pclib/ifcs/ValRdyBundle.py#L28

```
1:  def to_str( self, msg=None ):
2:   msg = self.msg if None else msg
3:   return valrdy_to_str( msg, self.val, self.rdy )
```

Shouldn't line  `2` be :

```
msg = self.msg if msg is None else msg
```
Thanks

cbatten, 2019-10-25:
Hmmm ... I think so? Note that pymtl2 will be deprecated this fall. We are working on pymtl3 -- a brand new and much better update!

 - https://github.com/cornell-brg/pymtl3

egaburov, 2019-10-25:
Great. Will models written for pymtl2 be compatible with pymtl3? If not, is there a porting guide from pymtl2 -> pymtl3. Thanks

cbatten, 2019-10-25:
They will not be compatible out of the box but the translation should be relatively straightforward. We are looking forward to the official pymtl3 3.0 release later this fall.