cornell-brg/pymtl #185: a small confusion of other language model support

hsqforfun, 2020-02-07:
Hi,
I notice that in PyMTL2, there are some systemc and verilog integration mechanism. But in the new PyMTL3, perhaps I only find systemverilog code support for integration. Is that true? I have a little confusion about why not support systemc model.
Thank you!

cbatten, 2020-02-07:
There is no reason PyMTL3 could not also support SystemC integration. we don’t really use SystemC in our research so this is not a high priority for us, but pull requests welcome!

closed 2020-02-07 by cbatten

jsn1993, 2020-02-07:
> Hi,
> I notice that in PyMTL2, there are some systemc and verilog integration mechanism. But in the new PyMTL3, perhaps I only find systemverilog code support for integration. Is that true? I have a little confusion about why not support systemc model.
> Thank you!

We have done some work to port systemc to pymtl3. It will probably be upstream in the near future.

hsqforfun, 2020-02-07:
I get it! Thank you!

cbatten, 2020-02-07:
I spoke too soon! @jsn1993 is the lead PyMTL3 developer ... if he says SystemC integration in PyMTL3 might be coming soon let's hold him to it :)